ayushsharma82/ElegantOTA #184: OTA crash when trying to upload

Tsjippy, 2024-05-03:
When trying to do an async ota of an esp8266 device powered via serial it fails: see logs below
```
Soft WDT reset

Exception (4):
epc1=0x401005f2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffea30 end: 3fffffb0 offset: 0160
3fffeb90:  3fff03d4 0000af4f 00000c9b 00001388  
3fffeba0:  00001388 00001388 0000a862 4020fc3d  
3fffebb0:  00000000 0000a862 e51eb851 00c2b258  
3fffebc0:  3fffec10 40203e50 00000034 0000a862  
3fffebd0:  00001388 3fff2e9c 3ffefefc 40203f43  
3fffebe0:  3fff26d4 00000350 3ffe864c 3fff2230  
3fffebf0:  3fff2e9c 00000020 00000020 3ffefdf4  
3fffec00:  00001f90 40203e50 3ffefefc 402035e0  
3fffec10:  402137a0 c800a8c0 402137a0 c800a8c0  
3fffec20:  3ffe85e0 00000350 3ffe864c 40100aaa  
3fffec30:  00000000 3ffefefc ffffffff 4020a954  
3fffec40:  00002248 00000449 3ffe864c 40100aaa  
3fffec50:  00000000 00000000 3fff2fdc 3ffe9a5c  
3fffec60:  3fffeccc 00000020 3fff2e9c 40100df8  
3fffec70:  00000000 00000001 3fffec90 3fff2230  
3fffec80:  3fffeccc 3ffefdf4 3fffecc0 40201740  
3fffec90:  00000000 0012001f 00000000 4020e8b5  
3fffeca0:  3fff02b4 3fff2f84 3fffeccc 4020ea1c  
3fffecb0:  3fff02b4 00000001 3fffed00 4020182c  
3fffecc0:  00000000 004e004f 00000000 3fff2fdc  
3fffecd0:  004e004f 80000000 00000000 0046004f  
3fffece0:  00000000 3fff2f84 004e004f 80000000  
3fffecf0:  3fff02b4 00000000 3fff256c 402018a5  
3fffed00:  3fff2f0c 0013001f 80000000 4020e731  

(...)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00057c40
~ld
����8�n��>r���n�b�l��l`ĜĒ����b�r���n�#ܸ��r�����8��� 1��Number of networks found: 2
Network name: Harmsen_EXT
Signal strength: -81
```

mathieucarbou, 2024-05-03:
seems to be an access to a null reference or dereferenced pointer.
You need to use platformio stack decoder option or arduino IDE to see the stack because the numbers here are all linked ot your compiled firmware so nobody will be able to help you.

Tsjippy, 2024-05-03:
thank you for you response.
I have the same issue with the OTA Async demo.

It uploads to 30% then crashes and reboots

I use ARDUINO IDE, this is the serial output

closed 2024-07-06 by github-actions[bot]