ayushsharma82/ElegantOTA #190: EspressifSystems version 3.0.1 arduino ide 1.8.19

Bredahl, 2024-06-05:
boards manager updated esp32 by EspressifSystems version 3.0.1
I get this error

In file included from C:\Users\sjjb\Desktop\DataLogLoraGaugesFreqC3\DataLogLoraGaugesFreqC3.ino:12:
C:\Users\sjjb\Documents\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:18:14: fatal error: esp_int_wdt.h: No such file or directory
   18 |     #include "esp_int_wdt.h"

mathieucarbou, 2024-06-05:
FYI the CI pass with Arduino 3.0.1, and I am using it in my project also, it works.
- https://github.com/ayushsharma82/ElegantOTA/pull/191
- https://github.com/mathieucarbou/ayushsharma82-ElegantOTA/actions/runs/9385148190

Bredahl, 2024-06-05:
> FYI the CI pass with Arduino 3.0.1, and I am using it in my project also, it works.
> 
> * [Arduino 3.0.1 #191](https://github.com/ayushsharma82/ElegantOTA/pull/191)
> * https://github.com/mathieucarbou/ayushsharma82-ElegantOTA/actions/runs/9385148190

If i go back to version 2.0.17 no problems.
don't know if it's because the chip is an esp32 C3

ayushsharma82, 2024-06-05:
@Bredahl Maybe because you are using "AsyncElegantOTA" instead of this repo?

Bredahl, 2024-06-05:
> @Bredahl Maybe because you are using "AsyncElegantOTA" instead of this repo?

Sorry my mistake

closed 2024-06-05 by Bredahl