ayushsharma82/ElegantOTA #193: Did you remove the Coffee Crap?

mallory98e, 2024-06-10:
Did you remove a buy me a coffee bloatware from this version? 
Cause it's intentionally obfuscated in the older one as well, I can hardly find where it loads it in.

Instead of putting in useless features like this nobody asked for:
[image]

You could add a file explorer to list out the SPIFFS...

I don't know where the original v1 uploading files to but they don't work the way as the Arduino Data Uploader. I would upload favicon.ico and some jpegs to my webserver.

ayushsharma82, 2024-06-10:
Hi @mallory98e 

Looks like you are coming from a time-machine haha. As the project matured, BMC was removed in v3.

Since v3, the project has been divided into OSS and Pro variants.  You are free to purchase the Pro version and support the project to continue experiencing "crap-free" software.

Unfortunately, making ElegantOTA a file manager software is something which is not planned. ElegantOTA will stick to only doing what it's best at - OTA updates.

I'd suggest updating your project to latest version and experience the refinements yourself.

mallory98e, 2024-06-11:
Dude I'm happy to update if there is improvement but why is there no SPIFFS file uploader?!
Where does it upload regular files then?

mathieucarbou, 2024-06-11:
> Dude I'm happy to update if there is improvement but why is there no SPIFFS file uploader?! Where does it upload regular files then?

ESP-DASH never included a file manager. It was coming from one of its dependency: ESPAsyncWebServer.

If you need it, you can still grab the 2 SPIFFS source files from https://github.com/yubox-node-org/ESPAsyncWebServer/blob/yuboxfixes-0xFEEDC0DE64-cleanup/src/SPIFFSEditor.h

SPIFFSEditor (this is how they called it) was removed from the maintained fork of ESPAsyncWebServer (https://github.com/mathieucarbou/ESPAsyncWebServer) for many reasons:

- SPIFFS is now obsoleted in favor of Little FS dues to file corruption issues
- It was also removed in order to reduce the library size and features to maintain
- It was not commonly used and for people who still need it, can be added back as a per case basis
- It was spawning an non configurable endpoint on the web server which could conflict with some user defined routes

I strongly suggest you to switch to LittleFS and a descent file manager

mathieucarbou, 2024-06-11:
@ayushsharma82 : a file manager would be a nice project idea by the way ;-)

mallory98e, 2024-06-13:
Lol and now it seems the old library interferes with the new one:

```
In file included from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/HTTP_Method.h:4,
                 from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/WebServer.h:30,
                 from /home/user/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:73,
                 from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:34:
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:95:6: error: 'HTTP_DELETE' conflicts with a previous declaration
   XX(0,  DELETE,      DELETE)       \
      ^
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:33:
/home/user/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
   HTTP_DELETE  = 0b00000100,
   ^~~~~~~~~~~
In file included from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/HTTP_Method.h:4,
                 from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/WebServer.h:30,
                 from /home/user/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:73,
                 from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:34:
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:96:6: error: 'HTTP_GET' conflicts with a previous declaration
   XX(1,  GET,         GET)          \
      ^
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:33:
/home/user/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:62:3: note: previous declaration 'WebRequestMethod HTTP_GET'
   HTTP_GET     = 0b00000001,
   ^~~~~~~~
In file included from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/HTTP_Method.h:4,
                 from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/WebServer.h:30,
                 from /home/user/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:73,
                 from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:34:
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:97:6: error: 'HTTP_HEAD' conflicts with a previous declaration
   XX(2,  HEAD,        HEAD)         \
      ^
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:33:
/home/user/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:67:3: note: previous declaration 'WebRequestMethod HTTP_HEAD'
   HTTP_HEAD    = 0b00100000,
   ^~~~~~~~~
In file included from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/HTTP_Method.h:4,
                 from /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WebServer/src/WebServer.h:30,
                 from /home/user/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:73,
                 from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:34:
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:98:6: error: 'HTTP_POST' conflicts with a previous declaration
   XX(3,  POST,        POST)         \
      ^
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
/arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from /tmp/arduino_modified_sketch_232777/AsyncDemo.ino:33:
/home/user/Arduino/libraries/ESPAsyncWebServer/src/ESPAsyncWebServer.h:63:3: note: previous declaration 'WebRequestMethod HTTP_POST'
   HTTP_POST    = 0b00000010,
[... 45 lines cut ...]
[... cut ...]
/home/user/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:107:10: note:   no known conversion for argument 1 from 'AsyncWebServer*' to 'WebServer*'
Multiple libraries were found for "WiFi.h"
 Used: /arduino/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WiFi
 Not used: /arduino/arduino-1.8.19/libraries/WiFi
exit status 1
no matching function for call to 'ElegantOTAClass::begin(AsyncWebServer*)'

```

Why didn't just call it the same name ... as in the Library it does show up with the same name.

ayushsharma82, 2024-06-13:
Read docs and changelog before you migrate to new 'major' number. You'll find your answer in the documentation website.

mallory98e, 2024-06-16:
Yes but is there any single reason on earth why should I upgrade? :rofl: 
Ok the coffee disappears a good one but how "better" that new version can get?!
It can upload a file that's it.

ayushsharma82, 2024-06-16:
Listen, I guess you don't know how it works everywhere, nobody is working for free to make you feel 'entitled'.

Use your own judgement here, If you think something is not working in your favour, you can:

1. Provide the repo with a fix/feature after consultation with maintainers.
2. Go ahead and create your own thing according to what you like.
3. Come to me with a requirement and we can deal it as a business.

You've constantly shown no regard for the maintainers and contributors through your comments, therefore I'm closing this issue as this is not a place to cast your opinions, you can open an issue anytime when you have a bug or feature request which aligns with the goal of the project.

closed 2024-06-16 by ayushsharma82