ayushsharma82/ElegantOTA #197: error after enable async mode

HARIS4820, 2024-06-27:
i am using arduino 2.3.2 appimage in ubuntu 24.4. installed esp32 board 3.0.2 by espressif systems. board doit esp32 devkit v1 .
[image]
after enable async mode i am getting this error.
[image]

Daryavahus, 2024-06-29:
I have similar error:
d:\Moje Dokumenty\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp: In lambda function:
d:\Moje Dokumenty\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:22:117: error: no matching function for call to 'AsyncWebServerRequest::beginResponse(int, const char [10], const uint8_t [10214], unsigned int)'
   22 |       AsyncWebServerResponse *response = request->beginResponse(200, "text/html", ELEGANT_HTML, sizeof(ELEGANT_HTML));
      |                                                                                                                     ^
In file included from d:\Moje Dokumenty\Arduino\libraries\ElegantOTA\src\ElegantOTA.h:52,
                 from d:\Moje Dokumenty\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:1:
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:246:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(int, const String&, const String&)'
  246 |     AsyncWebServerResponse *beginResponse(int code, const String& contentType=String(), const String& content=String());
      |                             ^~~~~~~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:246:29: note:   candidate expects 3 arguments, 4 provided
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:247:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(AsyncWebServerRequest::FS&, const String&, const String&, bool, AwsTemplateProcessor)'
  247 |     AsyncWebServerResponse *beginResponse(FS &fs, const String& path, const String& contentType=String(), bool download=false, AwsTemplateProcessor callback=nullptr);
      |                             ^~~~~~~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:247:47: note:   no known conversion for argument 1 from 'int' to 'AsyncWebServerRequest::FS&' {aka 'fs::FS&'}
  247 |     AsyncWebServerResponse *beginResponse(FS &fs, const String& path, const String& contentType=String(), bool download=false, AwsTemplateProcessor callback=nullptr);
      |                                           ~~~~^~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:248:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(AsyncWebServerRequest::File, const String&, const String&, bool, AwsTemplateProcessor)'
  248 |     AsyncWebServerResponse *beginResponse(File content, const String& path, const String& contentType=String(), bool download=false, AwsTemplateProcessor callback=nullptr);
      |                             ^~~~~~~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:248:48: note:   no known conversion for argument 1 from 'int' to 'AsyncWebServerRequest::File' {aka 'fs::File'}
  248 |     AsyncWebServerResponse *beginResponse(File content, const String& path, const String& contentType=String(), bool download=false, AwsTemplateProcessor callback=nullptr);
      |                                           ~~~~~^~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:249:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(Stream&, const String&, size_t, AwsTemplateProcessor)'
  249 |     AsyncWebServerResponse *beginResponse(Stream &stream, const String& contentType, size_t len, AwsTemplateProcessor callback=nullptr);
      |                             ^~~~~~~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:249:51: note:   no known conversion for argument 1 from 'int' to 'Stream&'
  249 |     AsyncWebServerResponse *beginResponse(Stream &stream, const String& contentType, size_t len, AwsTemplateProcessor callback=nullptr);
      |                                           ~~~~~~~~^~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:250:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(const String&, size_t, AwsResponseFiller, AwsTemplateProcessor)'
  250 |     AsyncWebServerResponse *beginResponse(const String& contentType, size_t len, AwsResponseFiller callback, AwsTemplateProcessor templateCallback=nullptr);
      |                             ^~~~~~~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ESPAsyncWebSrv\src/ESPAsyncWebSrv.h:250:100: note:   no known conversion for argument 3 from 'const uint8_t [10214]' {aka 'const unsigned char [10214]'} to 'AwsResponseFiller' {aka 'std::function<unsigned int(unsigned char*, unsigned int, unsigned int)>'}
  250 |     AsyncWebServerResponse *beginResponse(const String& contentType, size_t len, AwsResponseFiller callback, AwsTemplateProcessor templateCallback=nullptr);
      |                                                                                  ~~~~~~~~~~~~~~~~~~^~~~~~~~
d:\Moje Dokumenty\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp: In lambda function:
d:\Moje Dokumenty\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:210:51: warning: unused parameter 'filename' [-Wunused-parameter]
  210 |     }, [&](AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
      |                                            ~~~~~~~^~~~~~~~

version 3.1.1 works fine.

mathieucarbou, 2024-06-29:
you need to update your dependencies, especially ESPAsyncWebServer.

HARIS4820, 2024-06-29:
I use https://github.com/me-no-dev/ESPAsyncWebServer this library. My issue resolved

mathieucarbou, 2024-06-29:
FYI - see release notes of https://github.com/mathieucarbou/ESPAsyncWebServer/releases/ (which is the now maintained fork of [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer))

Daryavahus, 2024-06-30:
... and it works fine. Thank You

antwal, 2024-06-30:
Updated library but i get build error:

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01.html
PLATFORM: Espressif 8266 (3.2.0) > Espressif Generic ESP8266 ESP-01 512k
HARDWARE: ESP8266 80MHz, 80KB RAM, 512KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.1.0
|-- DHT sensor library @ 1.4.6
|-- ElegantOTA @ 3.1.2
|-- ESP Async WebServer @ 3.0.3
|-- LittleFS @ 0.1.0
|-- Adafruit Unified Sensor @ 1.1.14
|-- ESP8266WiFi @ 1.0
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- Wire @ 1.0
Building in release mode
Compiling .pio/build/esp01/src/main.cpp.o
Generating LD script .pio/build/esp01/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/esp01/lib703/Adafruit Unified Sensor/Adafruit_Sensor.cpp.o
Compiling .pio/build/esp01/lib7f2/DHT sensor library/DHT.cpp.o
Compiling .pio/build/esp01/lib7f2/DHT sensor library/DHT_U.cpp.o
Compiling .pio/build/esp01/lib0a8/ESPAsyncTCP-esphome/AsyncPrinter.cpp.o
Compiling .pio/build/esp01/lib0a8/ESPAsyncTCP-esphome/ESPAsyncTCP.cpp.o
Compiling .pio/build/esp01/lib0a8/ESPAsyncTCP-esphome/ESPAsyncTCPbuffer.cpp.o
Compiling .pio/build/esp01/lib0a8/ESPAsyncTCP-esphome/SyncClient.cpp.o
Compiling .pio/build/esp01/lib0a8/ESPAsyncTCP-esphome/tcp_axtls.c.o
Compiling .pio/build/esp01/libe07/Hash/Hash.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFiGratuitous.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Archiving .pio/build/esp01/lib703/libAdafruit Unified Sensor.a
Indexing .pio/build/esp01/lib703/libAdafruit Unified Sensor.a
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Archiving .pio/build/esp01/lib7f2/libDHT sensor library.a
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Indexing .pio/build/esp01/lib7f2/libDHT sensor library.a
Archiving .pio/build/esp01/libe07/libHash.a
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/WiFiClient.cpp.o
Indexing .pio/build/esp01/libe07/libHash.a
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/WiFiServer.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/WiFiUdp.cpp.o
Compiling .pio/build/esp01/lib0ee/ESP8266WiFi/enable_wifi_at_boot_time.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:141:57: warning: 'void AsyncWebServerRequest::send_P(int, const String&, const char*, AwsTemplateProcessor)' is deprecated: Replaced by send(...) [-Wdeprecated-declarations]
  141 |     request->send_P(200, "text/plain", String(t).c_str());
      |                                                         ^
In file included from src/main.cpp:18:
.pio/libdeps/esp01/ESP Async WebServer/src/ESPAsyncWebServer.h:308:10: note: declared here
  308 |     void send_P(int code, const String& contentType, PGM_P content, AwsTemplateProcessor callback = nullptr) {
      |          ^~~~~~
src/main.cpp: In lambda function:
src/main.cpp:146:57: warning: 'void AsyncWebServerRequest::send_P(int, const String&, const char*, AwsTemplateProcessor)' is deprecated: Replaced by send(...) [-Wdeprecated-declarations]
[... 36 lines cut ...]
Archiving .pio/build/esp01/lib0a4/libWire.a
Indexing .pio/build/esp01/lib0a4/libWire.a
Compiling .pio/build/esp01/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/esp01/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/esp01/FrameworkArduino/LwipDhcpServer-NonOS.cpp.o
Compiling .pio/build/esp01/FrameworkArduino/LwipDhcpServer.cpp.o
Compiling .pio/build/esp01/FrameworkArduino/LwipIntf.cpp.o
Compiling .pio/build/esp01/FrameworkArduino/LwipIntfCB.cpp.o
Compiling .pio/build/esp01/FrameworkArduino/MD5Builder.cpp.o
.pio/libdeps/esp01/ElegantOTA/src/ElegantOTA.cpp: In lambda function:
.pio/libdeps/esp01/ElegantOTA/src/ElegantOTA.cpp:76:71: error: 'FS_end' was not declared in this scope; did you mean 'va_end'?
   76 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                       ^~~~~~
      |                                                                       va_end
.pio/libdeps/esp01/ElegantOTA/src/ElegantOTA.cpp:76:88: error: 'FS_start' was not declared in this scope; did you mean 'va_start'?
   76 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                                        ^~~~~~~~
      |                                                                                        va_start
Compiling .pio/build/esp01/FrameworkArduino/Print.cpp.o
Archiving .pio/build/esp01/libd64/libLittleFS.a
Indexing .pio/build/esp01/libd64/libLittleFS.a
Compiling .pio/build/esp01/FrameworkArduino/Schedule.cpp.o
Compiling .pio/build/esp01/FrameworkArduino/StackThunk.cpp.o
*** [.pio/build/esp01/lib9cc/ElegantOTA/ElegantOTA.cpp.o] Error 1
```

mathieucarbou, 2024-06-30:
@antwal : please see the release notes of the dependency: https://github.com/mathieucarbou/ESPAsyncWebServer/releases

antwal, 2024-06-30:
@mathieucarbou 
> https://github.com/mathieucarbou/ESPAsyncWebServer/releases

this is not the problem, the library has been updated and the release indicates that send_P is deprecated

the problem is this, which generates an error and not a warning
```
.pio/libdeps/esp01/ElegantOTA/src/ElegantOTA.cpp: In lambda function:
.pio/libdeps/esp01/ElegantOTA/src/ElegantOTA.cpp:76:71: error: 'FS_end' was not declared in this scope; did you mean 'va_end'?
   76 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                       ^~~~~~
      |                                                                       va_end
.pio/libdeps/esp01/ElegantOTA/src/ElegantOTA.cpp:76:88: error: 'FS_start' was not declared in this scope; did you mean 'va_start'?
   76 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                                        ^~~~~~~~
      |                                                                                        va_start
```

I also include the platformio:

```
[env:esp01]
platform = espressif8266
board = esp01
framework = arduino
monitor_speed = 115200
board_build.filesystem = littlefs
build_flags = -DELEGANTOTA_USE_ASYNC_WEBSERVER=1
lib_deps = 
	bblanchon/ArduinoJson@^7.0.4
	adafruit/DHT sensor library@^1.4.6
	ayushsharma82/ElegantOTA@^3.1.2
	mathieucarbou/ESP Async WebServer@^3.0.3
```

ayushsharma82, 2024-06-30:
@antwal Can you try updating your ESP8266 platform in PIO? maybe that will fix it.

antwal, 2024-06-30:
@ayushsharma82 yes I updated to version 4.2.1, with the 3.2.0 I had it gave an error, now it works

ayushsharma82, 2024-06-30:
Closing this issue as solved.

closed 2024-06-30 by ayushsharma82