ayushsharma82/ElegantOTA #201: beginResponse_P - Braking Change in mathieucarbou / ESPAsyncWebServer for ESP32 in release 3.0.2

hagre, 2024-06-28:
see: https://github.com/mathieucarbou/ESPAsyncWebServer/pull/39

The beginResponse_P is only avaliable for ESP8266 at release 3.0.2

ayushsharma82, 2024-06-30:
@mathieucarbou  Always keep backwards compatibility. :) Too many little microcontrollers rest on your shoulders now.

ayushsharma82, 2024-07-06:
@sachiger as mathieu mentioned, have you updated your AsyncWebServer dependency to version >= 3.0.6?

sachiger, 2024-07-06:
I have ESPAsyncWebServer by Iacamera 3.1.0 installed.
I am not sure if I need to do anything else
Thanks

ayushsharma82, 2024-07-06:
@sachiger That’s not our recommended dependency: https://docs.elegantota.pro/async-mode/#dependencies

sachiger, 2024-07-07:
Hi,
I replaced ESPAsyncWebServer to 3.0.6
The web server (application) stopped responding, unless the HTML page is
very simple
with ESPAsyncWebServer 3.1.0 and ElegantOTA 3.1.1 all works fine.
Any ideas?
Thanks

mathieucarbou, 2024-07-07:
> Hi, I replaced ESPAsyncWebServer to 3.0.6 The web server (application) stopped responding, unless the HTML page is very simple with ESPAsyncWebServer 3.1.0 and ElegantOTA 3.1.1 all works fine. Any ideas? Thanks
> […](#)

Maybe clan all your libraries. You must have conflicts. I guess you are using Arduino IDE... FYI ESPAsyncWebServer 3.1.0 does not exist: this is a guy who released original repo under a new version, which under some case might cause a mess.

https://github.com/arduino/library-registry/pull/4585#

I suggest you  cleanup the dependencies, or use a real dev environment like Vscode with PlatformIO who are able to correctly deal with dependencies and versions.

yesnoj, 2024-07-08:
On my ESP32 S3 the board keeps crashing at boot.
Don't know why, i've used the correct dependencies, installing the .zip file with Arduino IDE.
The option to use another IDE is not contempled, as the project is quite finished.
Don't know if the LVLG stuff can , in someway, cause some issues, the error is the following :
"22:01:04.359 -> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled."

The code is super simple...
```
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ElegantOTA.h>

void connectOta(){
  if (!WiFi.softAP(ssid, password)) {
    LV_LOG_USER("Soft AP creation failed.");
    while(1);
  }
  IPAddress myIP = WiFi.softAPIP();
  LV_LOG_USER("AP IP address: %s",myIP.toString());
 

  ElegantOTA.setAutoReboot(false);
  ElegantOTA.setAuth(user, pass);
  ElegantOTA.begin(&otaServer);
  otaServer.begin();
  LV_LOG_USER("Server started");

  // ElegantOTA callbacks
  ElegantOTA.onStart(onOTAStart);
  ElegantOTA.onProgress(onOTAProgress);
  ElegantOTA.onEnd(onOTAEnd);
}

in the loop there is :   ElegantOTA.loop();

Any suggestion!? :/

sachiger, 2024-07-19:
> arduino/library-registry#4585

I tried to work with VScode/PlatformIO but failed to configure it right.
Working on Arduino IDE esphome-ESPAsyncTCP 2.0.0 failed to install,
collapsing all dependencies. Any solution/reason why it wouldn't install?
Many thanks
Sachi

mathieucarbou, 2024-07-19:
@sachiger : you can take a look at the pio file in this project as a start exemple. You can have pio compile your ino file.
This is how this project examples can be locally built.

sachiger, 2024-07-19:
Thank you for your prompt response. Where Can I find this PIO file?

mathieucarbou, 2024-07-19:
> Thank you for your prompt response. Where Can I find this PIO file?

In this project: `platformio.ini`

ayushsharma82, 2024-08-05:
Incase someone wants to use some other fork of ESPAsyncWebServer: https://docs.elegantota.pro/async-mode/#use-of-unrecommended-dependencies-experimental

closed 2024-08-05 by ayushsharma82