ayushsharma82/ElegantOTA #203: Platformio - ElegantOTA AsyncTCP library multiple defines clash with all versions of AsyncMQTT

ronsegal, 2024-07-03:
Under platformio arduino framework I cannot find a version of AsyncMQTT (tried all sorts of combinations) that works with ElegantOTA without a library clash that results in multiple define linking errors for AsyncTCP.  The mathieucarbou/Async TCP @ ^3.1.4 library required by ElegantOTA isn't directly available in platformio registry, although that might not help anyway. There was no problem with the previous AsyncElegantOTA.  Therefore currently I am having to use the old AsyncElegantOTA as a workaround.

mathieucarbou, 2024-07-03:
>  I cannot find a version of AsyncMQTT 

Are you talking about this one ? https://github.com/marvinroger/async-mqtt-client

FYI it is quite old, now the ESP-IDF includes an async MQTT client so no need to add another one. 
You can look at this lib which is an easy to use wrapper around the ESP-IDF API: https://github.com/mathieucarbou/MycilaMQTT

> library clash that results in multiple define linking errors

Probably because the async mqtt client you are using points to an old AsyncTCP dependency. The one used through this project is improved and API compatible, see the changes: https://github.com/mathieucarbou/AsyncTCP.

You just need to edit the dependencies of the mqtt client you are using (library.json), or remove them. Or use the mqtt api included now in Arduino / ESP-IDF 5.x.

> The mathieucarbou/Async TCP @ ^3.1.4 library required by ElegantOTA isn't directly available in platformio registry

It is: https://registry.platformio.org/libraries/mathieucarbou/Async%20TCP

ronsegal, 2024-07-03:
Thanks for getting back on this Mathieu.

Apologies to all that on further investigation this is a non issue.

Had mistyped the name of the mathieucarbou AsyncTCP library so was unable to specify properly in platformio.ini

With that done ElegantOTA and AsyncMqttClient build together
lib_deps = 
    mathieucarbou/Async TCP @ ^3.1.4
    heman/AsyncMqttClient-esphome@^2.1.0
    invzblio/SerialToWifi@^0.1.6
    ropg/ezTime@^0.8.3
    bblanchon/ArduinoJson@^7.1.0
    fbiego/ESP32Time@^2.0.6
    ayushsharma82/ElegantOTA@^3.1.2

Cheers
Ron

closed 2024-07-03 by ronsegal

mathieucarbou, 2024-07-03:
> heman/AsyncMqttClient-esphome@^2.1.0

I didn't know this one! That's nice that ESPHome folks are maintaining it.

But it is using `AsyncTCP-esphome` which has bugs inside. That's why I forked it and renamed it to `Async TCP`, closer to `Async TCP`

ESPHome guys also tend to rename the library they fork.

ronsegal, 2024-07-03:
There's an AsyncTCP-esphome 2.1.3, which is 5 months old, with another version, same name, 3 years old.  So perhaps the newer version is debugged!

mathieucarbou, 2024-07-03:
> There's an AsyncTCP-esphome 2.1.3, which is 5 months old, with another version, same name, 3 years old. So perhaps the newer version is debugged!

I know. I've opened some PRs (https://github.com/esphome/AsyncTCP/pulls).
But there's ni activity.
They mainly forked in order to be able to point the libraries the maintain elsewhere to it.