ayushsharma82/ElegantOTA #206: ElegantOTA PlatformIO error

Bredahl, 2024-07-11:
I get these two error messages,
with query parameters

a value of type "const AsyncWebParameter *" cannot be used to initialize an entity of type "AsyncWebParameter *"
invalid conversion from 'const AsyncWebParameter*' to 'AsyncWebParameter*' [-fpermissive]

when I use this code in Platform IO

   server.on("/itvl", HTTP_GET, [](AsyncWebServerRequest *request){
    int paramsNr = request->params();
    Serial.println(paramsNr);
 
    for(int i=0;i<paramsNr;i++){
         
        AsyncWebParameter* p = request->getParam(i);
        Serial.print("Param name: ");
        Serial.println(p->name());
        Serial.print("Param value: ");
        Serial.println(p->value());
        Serial.println("------");
    }
     
    request->send(200, "text/plain", "message received");
  });

ayushsharma82, 2024-07-11:
This is not related to ElegantOTA.

Bredahl, 2024-07-11:
No, but it is related to those libraries
you recommend AsyncTCP

Bredahl, 2024-07-11:
for Esp32
(mathieucarbou) [AsyncTCP](https://github.com/mathieucarbou/AsyncTCP#v3.1.4) @ v3.1.4
(mathieucarbou) [ESPAsyncWebServer](https://github.com/mathieucarbou/ESPAsyncWebServer#v3.0.5) - v3.0.5

ayushsharma82, 2024-07-11:
Then please open the issue in repo for dependencies. Mathieu will help you, those forks are not maintained by me.

mathieucarbou, 2024-07-11:
@Bredahl : the api was fixed. The object is not mutable so you need to use const. The api enforces that now 

'const AsyncWebParameter* p = request->getParam(i);'

Bredahl, 2024-07-11:
thank you very much.
That's what made it work!!

closed 2024-07-11 by Bredahl