ayushsharma82/ElegantOTA #208: Compilation error on ESP32-S3

plystdr, 2024-07-19:
Hi there,

For some reason the code won't compile and throw error related with AsyncTCP_RP2040.
However, I am using ESP32-S3 board and nowhere in my code it is defined as RP2040.

I am on platformio with esp-idf and arduino as a component.

```
In file included from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:104:
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
In file included from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncPrinter.h:51,
                 from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncPrinter.cpp:51:
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
In file included from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_buffer.h:60,
                 from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_buffer.cpp:58:
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj.c.o
In file included from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/SyncClient.cpp:53:
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj_class.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj_draw.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj_pos.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj_scroll.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj_style.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj_style_gen.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_obj_tree.c.o
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:331:8: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
   addr.addr = ip;
        ^~~~
        u_addr
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:390:42: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
     returnValue = connect(IPAddress(addr.addr), port);
                                          ^~~~
                                          u_addr
In file included from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:109,
                 from .pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:104:
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:393:73: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
     ATCP_LOGDEBUG3("connect: dns_gethostbyname => IP =", IPAddress(addr.addr), ", returnValue = ", returnValue ? "TRUE" : "FALSE");
                                                                         ^~~~
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_Debug.h:87:128: note: in definition of macro 'ATCP_LOGDEBUG3'
 #define ATCP_LOGDEBUG3(x,y,z,w)  if(_ASYNCTCP_RP2040W_LOGLEVEL_>3) { ATCP_PRINT_MARK; ATCP_PRINT(x); ATCP_PRINT_SP; ATCP_PRINT(y); ATCP_PRINT_SP; ATCP_PRINT(z); ATCP_PRINT_SP; ATCP_PRINTLN(w); }
                                                                                                                                ^
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::operator==(const AsyncClient&)':
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:468:66: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
   return (_pcb != NULL && other._pcb != NULL && (_pcb->remote_ip.addr == other._pcb->remote_ip.addr) &&
                                                                  ^~~~
                                                                  u_addr
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:468:96: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_refr.c.o
   return (_pcb != NULL && other._pcb != NULL && (_pcb->remote_ip.addr == other._pcb->remote_ip.addr) &&
                                                                                                ^~~~
                                                                                                u_addr
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'void AsyncClient::_dns_found(ip_addr_t*)':
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:1027:26: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
     connect(IPAddress(p->addr), _connect_port);
                          ^~~~
                          u_addr
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'uint32_t AsyncClient::getRemoteAddress()':
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:1236:26: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
   return _pcb->remote_ip.addr;
                          ^~~~
                          u_addr
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'uint32_t AsyncClient::getLocalAddress()':
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:1256:25: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
   return _pcb->local_ip.addr;
                         ^~~~
                         u_addr
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'void AsyncServer::begin()':
.pio/libdeps/demotest/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:1721:14: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
   local_addr.addr = (uint32_t) _addr;
              ^~~~
              u_addr
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\core\lv_theme.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\draw\arm2d\lv_gpu_arm2d.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\draw\lv_draw.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\draw\lv_draw_arc.c.o
Compiling .pio\build\demotest\lib8e0\lvgl-8.3.11\draw\lv_draw_img.c.o
*** [.pio\build\demotest\lib40f\AsyncTCP_RP2040W\AsyncPrinter.cpp.o] Error 1
*** [.pio\build\demotest\lib40f\AsyncTCP_RP2040W\AsyncTCP_RP2040W.cpp.o] Error 1
*** [.pio\build\demotest\lib40f\AsyncTCP_RP2040W\AsyncTCP_RP2040W_buffer.cpp.o] Error 1
*** [.pio\build\demotest\lib40f\AsyncTCP_RP2040W\SyncClient.cpp.o] Error 1
```

ayushsharma82, 2024-07-19:
It's because ElegantOTA also support RP2040 and PIO pulls RP2040 dependency as well. To avoid that - follow steps in this: https://docs.netwizard.pro/docs/intro/installation#modifications-in-platform-io-ini

plystdr, 2024-07-19:
I followed the instructions. Added necessary settings to the platformio.ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[common]
lib_deps = https://github.com/joltwallet/esp_littlefs.git

[env:demoproject]
platform = espressif32
board = custom-circuit
board_build.mcu = esp32s3
framework = espidf, arduino
upload_protocol = esptool
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = partition_scheme.csv
board_build.filesystem = littlefs
lib_compat_mode = soft
lib_ldf_mode = chain
lib_deps = ${common.lib_deps}
extra_scripts =
	tasks.py
	;pre:preprocess.py
	;post:postprocess.py
build_flags = 
	-DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-DARDUINO_ESP32S3_DEV
	-DCONFIG_SPIRAM_USE
    -mfix-esp32-psram-cache-issue
	-w
	-Wno-unused-variable
	-Wno-unused-function
```

Cleaned the .pio directory.
Result still the same.

[image]

plystdr, 2024-07-19:
anyways, apparently manual installation causing some problems with platformio.
I ended up installing it from platformio libraries itself.
It looks ok now.

Thanks!

closed 2024-07-19 by plystdr