ayushsharma82/ElegantOTA #210: ElegantOTA in a ESP32 not compiling.

FOHJimJohnson, 2024-07-24:
I have a program that I've had running for over a year on an ESP8266. It uses Elegant OTA and works great. I've now purchased a ESP32 to run this program due to its increased "horsepower" I'm using PlatformIO in a Win10 machine. I've got the program running great with the exception of ElegantOTA. It gives me compile errors when I try to put it in. which is a bummer, these boards are around the house and going to each one th update is not fun. I hope you can help. Here is my ini:

[env:ESP32_Heatpump_Interface]
platform = espressif32
board = adafruit_feather_esp32_v2
board_build.filesystem = littlefs
board_build.partitions = default_8MB.csv
framework = arduino
lib_deps = 
          adafruit/Adafruit NeoPixel@^1.12.3
          https://github.com/me-no-dev/ESPAsyncWebServer.git
          https://github.com/ayushsharma82/ElegantOTA@^3.1.3
monitor_speed = 115200
build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1 -O2

Here is the Errors from the compiler:

.pio/libdeps/ESP32_Heatpump_Interface/ElegantOTA/src/ElegantOTA.cpp: In lambda function:
.pio/libdeps/ESP32_Heatpump_Interface/ElegantOTA/src/ElegantOTA.cpp:22:117: error: no matching function for call to 'AsyncWebServerRequest::beginResponse(int, const char [10], const uint8_t [10214], unsigned int)'
       AsyncWebServerResponse *response = request->beginResponse(200, "text/html", ELEGANT_HTML, sizeof(ELEGANT_HTML));
                                                                                                                     ^
In file included from .pio/libdeps/ESP32_Heatpump_Interface/ElegantOTA/src/ElegantOTA.h:68,
                 from .pio/libdeps/ESP32_Heatpump_Interface/ElegantOTA/src/ElegantOTA.cpp:1:
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:246:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(int, const String&, const String&)'
     AsyncWebServerResponse *beginResponse(int code, const String& contentType=String(), const String& content=String());
                             ^~~~~~~~~~~~~
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:246:29: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:247:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(AsyncWebServerRequest::FS&, const String&, const String&, bool, AwsTemplateProcessor)'
     AsyncWebServerResponse *beginResponse(FS &fs, const String& path, const String& contentType=String(), bool download=false, AwsTemplateProcessor callback=nullptr);
                             ^~~~~~~~~~~~~
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:247:29: note:   no known conversion for argument 1 from 'int' to 'AsyncWebServerRequest::FS&' {aka 'fs::FS&'}
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:248:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(AsyncWebServerRequest::File, const String&, const String&, bool, AwsTemplateProcessor)'
     AsyncWebServerResponse *beginResponse(File content, const String& path, const String& contentType=String(), bool download=false, AwsTemplateProcessor callback=nullptr);
                             ^~~~~~~~~~~~~
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:248:29: note:   no known conversion for argument 1 from 'int' to 'AsyncWebServerRequest::File' {aka 'fs::File'}
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:249:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(Stream&, const String&, size_t, AwsTemplateProcessor)'
     AsyncWebServerResponse *beginResponse(Stream &stream, const String& contentType, size_t len, AwsTemplateProcessor callback=nullptr);
                             ^~~~~~~~~~~~~
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:249:29: note:   no known conversion for argument 1 from 'int' to 'Stream&'
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:250:29: note: candidate: 'AsyncWebServerResponse* AsyncWebServerRequest::beginResponse(const String&, size_t, AwsResponseFiller, AwsTemplateProcessor)'
     AsyncWebServerResponse *beginResponse(const String& contentType, size_t len, AwsResponseFiller callback, AwsTemplateProcessor templateCallback=nullptr);
                             ^~~~~~~~~~~~~
.pio/libdeps/ESP32_Heatpump_Interface/ESP Async WebServer/src/ESPAsyncWebServer.h:250:29: note:   no known conversion for argument 3 from 'const uint8_t [10214]' {aka 'const unsigned char [10214]'} to 'AwsResponseFiller' {aka 'std::function<unsigned int(unsigned char*, unsigned int, unsigned int)>'}
*** [.pio\build\ESP32_Heatpump_Interface\lib0f9\ElegantOTA\ElegantOTA.cpp.o] Error 1

ayushsharma82, 2024-07-24:
@FOHJimJohnson  Dependencies were changed. We now use a fork of AsyncWebServer: https://docs.elegantota.pro/async-mode/#dependencies

FOHJimJohnson, 2024-07-24:
I'm stull a bit if a noob with PlatformIO, could you give me the correct Line(s) for the ini? I could seem to find it in the link you sent.
Thanks for a quick response.

FOHJimJohnson, 2024-07-24:
Nevermind, I figured it out! WooHoo! is is working great now.  Thanks!

closed 2024-07-24 by ayushsharma82