ayushsharma82/ElegantOTA #211: ESP-Home incompatibility

k4lls, 2024-07-27:
I have a project with esphome/AsyncTCP-esphome@2.1.3 and ottowinter/ESPAsyncWebServer-esphome @ ^3.2.2.
I can't use ElegantOTA with these library since you are using mathieu carbou's library. Sadly they are not fully compatible. Is there a way to allow the use of different library ?

mathieucarbou, 2024-07-27:
They are: our fork is based on ESPhome 's one.
Please open an issue in the relevant projet and show your compile issue I will have a look to help.
It is also possible that you are using deprecated methods for esp8266 in your code.

ayushsharma82, 2024-08-05:
@k4lls Just added with v3.1.4 - https://docs.elegantota.pro/async-mode/#use-of-unrecommended-dependencies-experimental

closed 2024-08-05 by ayushsharma82