ayushsharma82/ElegantOTA #218: [BUG] Build error for ESP8266 with latest version on PlatformIO

antwal, 2024-08-23:
Initially it worked until a few months ago, now that I have updated a bit everything in the compilation it generates a link error:

Error Build log [here](https://pastebin.com/iVMJdx5v)

if I remove all the elegantota, everything works correctly

Success Build log [here](https://pastebin.com/9jYvZDmc)

PlatformIO Config File:

```
[env:esp01]
platform = espressif8266
board = esp01_1m
framework = arduino
board_build.flash_mode = dout
monitor_speed = 115200
board_build.filesystem = littlefs
;build_flags = -DELEGANTOTA_USE_ASYNC_WEBSERVER=1
lib_deps = 
	bblanchon/ArduinoJson@^7.0.4
	adafruit/DHT sensor library@^1.4.6
	mathieucarbou/ESP Async WebServer@^3.0.3
;	ayushsharma82/ElegantOTA@^3.1.5
	ayushsharma82/WebSerial@^2.0.6
```

antwal, 2024-08-23:
The version 3.1.2 working, [build log](https://pastebin.com/VpRwQQXR)

mathieucarbou, 2024-08-23:
Please try with :

mathieucarbou/ESPAsyncWebServer @ 3.1.5

mathieucarbou, 2024-08-23:
In theory with pio you should not define the dependencies because they will be found by PIO in the elegant ota descriptor.

antwal, 2024-08-23:
> Please try with :
> 
> mathieucarbou/ESPAsyncWebServer @ 3.1.5

Resolving esp01 dependencies...
Library Manager: Installing mathieucarbou/ESP Async WebServer @ 3.1.5
UnknownPackageError: Could not find the package with 'mathieucarbou/ESP Async WebServer @ 3.1.5' requirements for your system 'darwin_x86_64'

mathieucarbou, 2024-08-23:
I wrote `ESPAsyncWebServer`, not `ESP Async WebServer`

antwal, 2024-08-23:
> mathieucarbou/ESPAsyncWebServer @ 3.1.5

I had already solved it in another way, but this works

closed 2024-08-23 by antwal