ayushsharma82/ElegantOTA #222: Async does not work, async example is not correct

taseraul, 2024-09-14:
While trying to compile the example using async i noticed it lacked the correct definition
The example is configured using the default behaviour which is not async
#define ELEGANTOTA_USE_ASYNC_WEBSERVER 0

`C:\Users\xxx\Documents\Arduino\libraries\ElegantOTA\examples\AsyncDemo\AsyncDemo.ino: In function 'void setup()':
C:\Users\xxx\Documents\Arduino\libraries\ElegantOTA\examples\AsyncDemo\AsyncDemo.ino:88:20: error: cannot convert 'AsyncWebServer*' to 'WebServer*'
88 |   ElegantOTA.begin(&server);    // Start ElegantOTA
|                    ^~~~~~~
|                    |
|                    AsyncWebServer*`

if i change the define then i will get compilation errors:

`c:/users/xxx/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\xxx\AppData\Local\Temp\arduino\sketches\1A557C2191BC4B134DE1E63A1B0A4A32\sketch\AsyncDemo.ino.cpp.o:(.literal._Z5setupv+0x64): undefined reference to `ElegantOTAClass::begin(AsyncWebServer*, char const*, char const*)'
c:/users/xxx/appdata/local/arduino15/packages/esp32/tools/esp-xs3/2302/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\xxx\AppData\Local\Temp\arduino\sketches\1A557C2191BC4B134DE1E63A1B0A4A32\sketch\AsyncDemo.ino.cpp.o: in function `setup()':
C:\Users\xxx\Documents\Arduino\libraries\ElegantOTA\examples\AsyncDemo/AsyncDemo.ino:89: undefined reference to `ElegantOTAClass::begin(AsyncWebServer*, char const*, char const*)'`

Tested on esp32 and esp32s3 with esp arduino core 3.0.4

KevinKadow, 2024-09-15:
Need to manually force async mode by editing the ElegantOTA.h header file, see https://docs.elegantota.pro/async-mode/

If you use the Arduino library manager to update the library, must go back in and manually re-apply the change.

taseraul, 2024-09-16:
```
#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1
#include "ElegantOTA.h"

... arduino sketch
```

 yes, i did change it, the 2nd set of errors come when i do that. I did put it right before the header include. which will overide the definition in the header file.

I tried now to force the define in the header file itself and it seems to be working. Probably the compiler first compiles the library and and takes the default definition and ignores the overwrite i have in the sketch.

closed 2024-09-16 by taseraul