ayushsharma82/ElegantOTA #225: Request using ESPAsyncWebServer > 3.3.0 with authentication middlewares

DRSDavidSoft, 2024-10-01:
Hi there @ayushsharma82,

Thanks for such an excellent library! I would like to request a small change in how the authentication (username and password is handled), particularly here:

https://github.com/ayushsharma82/ElegantOTA/blob/5117cab98e7068686e3007ac863c8c466f197d0f/src/ElegantOTA.cpp#L19-L21

If possible, can you migrate to the [new authentication middlware](https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#how-to-use-middleware)?

The benefit is that instead of relying on ElegantOTA to handle the authentication, we can delegate it to the ESPAsyncWebServer to handle all authentication.

I'm not confident enough to open a PR, so I would appreciate it if you consider this, thank you! 🤝

mathieucarbou, 2024-10-01:
@DRSDavidSoft : I am using ElegantOTA too, and what I do for ElegantOTA and WebSerial and ESP-DASH is to init. them without auth.

This way, I am controlling authc from the middleware added to the server.

But it only works of course if every endpoint are protected. If you only want to protected ElegantOTA ones, in this case, username/password has to be set.

Also, the way @ayushsharma82 has implemented auth (he is not calling `setAuthentication()` on the handler) is totally fine and will avoid a new internal middleware to be created per handler. But I agree with you this is less flexible since currently there is no way to control the auth method.

DRSDavidSoft, 2024-10-01:
That's true, I guess we could say ElegantOTA (and the other libs) are providing a secondary authentication mechanism, which can be used together with the middleware.

For more granular authentication control, the approach you are using is acceptable: instead of setting a username/password combo on the ElegantOTA, just attaching a custom middleware to control the `/update` endpoint.

A great usecase for this would be to be able to grant access to different user levels, e.g. allow the firmware to be updated based on only some user accounts rather than all authenticated users.

P.S. such a great library and I'm slightly surprised and so happy to see you also using it too @mathieucarbou 😁

mathieucarbou, 2024-10-01:
> P.S. such a great library and I'm slightly surprised and so happy to see you also using it too @mathieucarbou 😁

I love the UX from @ayushsharma82 : they are great. I've contributed a lot of code in his projects, especially in ESP-DAHS the last years: it now has a lot more features and perf improvements ;-)

I use ElegantOTA mostly in [MycilaSafeBoot](https://oss.carbou.me/MycilaSafeBoot/), which is a small recovery image allowing to have a huge app partition more than 3Mb, instead of being limited to 2Mb because of the dual-requirement of traditional OTA partition schemes.

ayushsharma82, 2024-10-22:
I guess I'll mark it completed 😄

closed 2024-10-22 by ayushsharma82