ayushsharma82/ElegantOTA #226: ElegentOTA does not build due to PICO 2040 dependency

UwUGirla, 2024-10-02:
**Compiling .pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\AsyncPrinter.cpp.o
Compiling .pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\AsyncTCP_RP2040W.cpp.o
Compiling .pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\AsyncTCP_RP2040W_buffer.cpp.o
Compiling .pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\SyncClient.cpp.o
In file included from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:104:
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
In file included from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_buffer.h:60,
                 from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_buffer.cpp:58:
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
In file included from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncPrinter.h:51,
                 from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncPrinter.cpp:51:
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
In file included from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.cpp:53:
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^~~~~
In file included from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/opt.h:51,
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip_addr.h:40,
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/tcpip_adapter/include/tcpip_adapter_types.h:18,
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/tcpip_adapter/include/tcpip_adapter.h:24,
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_netif/include/esp_netif.h:35,
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_event/include/esp_event_legacy.h:22,
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_event/include/esp_event.h:28,
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:27, 
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:28,     
                 from C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,        
                 from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:84,
                 from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.cpp:53:
C:/Users/Administrator/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/port/esp32/include/lwipopts.h:439: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_LWIP_TCP_MSS

In file included from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.h:60,
                 from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.cpp:52:
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/async_config.h:35: note: this is the location of the previous definition
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)

In file included from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.h:60,
                 from .pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.cpp:52:
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/async_config.h:35:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope  
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.h:86:34: note: in expansion of macro 'TCP_MSS'
     SyncClient(size_t txBufLen = TCP_MSS);
                                  ^~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/async_config.h:35:25: note: suggested alternative: 'CONFIG_LWIP_TCP_MSS'
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.h:86:34: note: in expansion of macro 'TCP_MSS'
     SyncClient(size_t txBufLen = TCP_MSS);
                                  ^~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/async_config.h:35:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope  
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.h:87:55: note: in expansion of macro 'TCP_MSS'
     SyncClient(AsyncClient *client, size_t txBufLen = TCP_MSS);
                                                       ^~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/async_config.h:35:25: note: suggested alternative: 'CONFIG_LWIP_TCP_MSS'
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/SyncClient.h:87:55: note: in expansion of macro 'TCP_MSS'
     SyncClient(AsyncClient *client, size_t txBufLen = TCP_MSS);
                                                       ^~~~~~~
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:331:8: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
[... 32 lines cut ...]
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:1236:26: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member 
named 'addr'; did you mean 'u_addr'?
   return _pcb->remote_ip.addr;
                          ^~~~
                          u_addr
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'uint32_t AsyncClient::getLocalAddress()':   
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:1256:25: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member 
named 'addr'; did you mean 'u_addr'?
   return _pcb->local_ip.addr;
                         ^~~~
                         u_addr
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'void AsyncServer::begin()':
.pio/libdeps/nodemcu-32s/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:1721:14: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member 
named 'addr'; did you mean 'u_addr'?
   local_addr.addr = (uint32_t) _addr;
              ^~~~
              u_addr
*** [.pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\AsyncPrinter.cpp.o] Error 1
*** [.pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\AsyncTCP_RP2040W_buffer.cpp.o] Error 1
*** [.pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\AsyncTCP_RP2040W.cpp.o] Error 1
*** [.pio\build\nodemcu-32s\lib711\AsyncTCP_RP2040W\SyncClient.cpp.o] Error 1**

[image]

[image]

mathieucarbou, 2024-10-02:
Hello,

You can try:

```ini
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#f2687073f73d554c9db41f29b4769fd9703f4e55
board = rpipicow

lib_compat_mode = strict
lib_ldf_mode = chain
lib_deps = mathieucarbou/ESPAsyncWebServer @ 3.3.7
```

I saw they added some new commits in rpi platform github repo which break everything, so sticking to `f2687073f73d554c9db41f29b4769fd9703f4e55` makes it work like before.

Maybe @ayushsharma82 would have some clues how to use the new master version. 
But the `platform-raspberrypi` repo not using tags make it quite unstable to use, except if you are fixing the version with a tag

UwUGirla, 2024-10-04:
Hello, I’m using just the same ESP32, on ESP8266 my assembly goes without any problems
[image]