ayushsharma82/ElegantOTA #228: cannot convert 'AsyncWebServer*' to 'WebServer*'

gjniewenhuijse, 2024-10-16:
At the Arduino ide 2.3.3. i received an error at compile, any idea why? Before it works for me.

C:\Temp\Bluetti_ESP32_zwembad\Bluetti_ESP32\BWifi.cpp: In function 'void initBWifi(bool)':
C:\Temp\Bluetti_ESP32_zwembad\Bluetti_ESP32\BWifi.cpp:585:22: error: cannot convert 'AsyncWebServer*' to 'WebServer*'
  585 |     ElegantOTA.begin(&server);
      |                      ^~~~~~~
      |                      |
      |                      AsyncWebServer*
In file included from C:\Temp\Bluetti_ESP32_zwembad\Bluetti_ESP32\BWifi.cpp:10:
c:\Users\gj.niewenhuijse\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:107:38: note:   initializing argument 1 of 'void ElegantOTAClass::begin(WebServer*, const char*, const char*)'
  107 |     void begin(ELEGANTOTA_WEBSERVER *server, const char * username = "", const char * password = "");
      |                                      ^
C:\Temp\Bluetti_ESP32_zwembad\Bluetti_ESP32\BWifi.cpp:587:22: error: cannot convert 'AsyncWebServer*' to 'WebServer*'
  587 |     ElegantOTA.begin(&server, wifiConfig.ota_username, wifiConfig.ota_password);
      |                      ^~~~~~~
      |                      |
      |                      AsyncWebServer*
c:\Users\gj.niewenhuijse\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:107:38: note:   initializing argument 1 of 'void ElegantOTAClass::begin(WebServer*, const char*, const char*)'
  107 |     void begin(ELEGANTOTA_WEBSERVER *server, const char * username = "", const char * password = "");
      |                                      ^

exit status 1

Compilation error: cannot convert 'AsyncWebServer*' to 'WebServer*'

DennisT2018, 2024-10-17:
I've seen this when the Async web server hasn't been enabled in the .h file. See docs here: https://docs.elegantota.pro/async-mode/

ayushsharma82, 2024-10-22:
Solution provided by @DennisT2018 . Closing the issue.

closed 2024-10-22 by ayushsharma82

juanmartin84, 2024-10-28:
Hello, i'm having this problem.

Tried to check the link provided by @DennisT2018 but seems broken.

[image]

ayushsharma82, 2024-10-28:
Docs were recently updated. Here’s the new link: https://docs.elegantota.pro/getting-started/async-mode