ayushsharma82/ElegantOTA #229: Progress bar always 100%

georgepapas123, 2024-10-22:
Progress bar jumps immediately to 100% and stays there until finished.
onOTAProgress() works properly showing steady increments until complete.
Why can't progress bar be updated in onOTAProgress() also?
Using Windows 10 22H2, Arduino IDE 2.3.3, ESP8266, EspAsyncWebServer 3.3.17, ElegantOTA 3.1.6 async mode

ayushsharma82, 2024-10-22:
@georgepapas123 It's an issue with apple devices (specifically how Safari caches requests) and there is no work around for it. You can try the same on any other browser and it will work fine.

ayushsharma82, 2024-10-22:
Just noticed you wrote "Windows". Are you trying to upload the bin file through windows itself or from any apple device?

georgepapas123, 2024-10-22:
The upload is done on Windows using Firefox 131.0.3 (64-bit).
You got me thinking and I tried Opera and it works so I guess chrome based browsers work.
It seems it's something in Firefox, but since a lot of people use Firefox if it's something simple it would be great to be fixed.

ayushsharma82, 2024-10-22:
Thanks for the insight. I'll give it a look soon if that can be fixed.

Usually the progress bar breaks because the browser is doing some sort of caching due to security concerns over HTTP protocol and it doesn't report back exact amount of bytes sent to server. There is no getting around it for the time being if this is the case (like Safari browser) because it's a browser level restriction.

georgepapas123, 2024-10-23:
For "problematic" browsers like Firefox, Safari etc. may I suggest a palindromic progress bar, going back and forth until completion, then you would just hide it and show the success (or failure) message.
This will be incredibly useful in systems where people now see a frozen progress bar and think OTA failed because they may not wait for about a minute or so to see the process complete.

ayushsharma82, 2024-10-24:
@georgepapas123  Good idea. I'll implement that.

closed 2025-02-12 by github-actions[bot]