ayushsharma82/ElegantOTA #235: Errors while linking async example on PlatformIO

pinomarcelo, 2024-11-17:
Sketch: AsyncDemo.ino
All recomendations where applied.

platformio.ini
[env:esp32doit-devkit-v1]
;[env:nodemcu-32s]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
lib_compat_mode = strict
lib_deps = 
	esphome/ESPAsyncWebServer-esphome@^3.3.0
	ayushsharma82/ElegantOTA@^3.1.6
build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1

Trying to compile sketch i get this:

Executing task in folder Prueba OTA ESP32: C:\Users\pinom\.platformio\penv\Scripts\platformio.exe run 

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.9.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- ElegantOTA @ 3.1.6
|-- ESPAsyncWebServer @ 3.3.22
|-- WiFi @ 2.0.0
Building in release mode
Linking .pio\build\esp32doit-devkit-v1\firmware.elf
c:/users/pinom/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\lib0c8\libESPAsyncWebServer.a(WebRequest.cpp.o): in function `AsyncWebServerRequest::_onPoll()':
E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer/src/WebRequest.cpp:155: multiple definition of `AsyncWebServerRequest::_onPoll()'; .pio\build\esp32doit-devkit-v1\lib9c8\libESPAsyncWebServer-esphome.a(WebRequest.cpp.o):E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer-esphome/src/WebRequest.cpp:190: first defined here
c:/users/pinom/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\lib0c8\libESPAsyncWebServer.a(WebRequest.cpp.o): in function `AsyncWebServerRequest::_onAck(unsigned int, unsigned int)':
E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer/src/WebRequest.cpp:170: multiple definition of `AsyncWebServerRequest::_onAck(unsigned int, unsigned int)'; .pio\build\esp32doit-devkit-v1\lib9c8\libESPAsyncWebServer-esphome.a(WebRequest.cpp.o):E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer-esphome/src/WebRequest.cpp:197: first defined here
c:/users/pinom/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\lib0c8\libESPAsyncWebServer.a(WebRequest.cpp.o): in function `AsyncWebServerRequest::_onError(signed char)':
E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer/src/WebRequest.cpp:185: multiple definition of `AsyncWebServerRequest::_onError(signed char)'; 
.pio\build\esp32doit-devkit-v1\lib9c8\libESPAsyncWebServer-esphome.a(WebRequest.cpp.o):E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer-esphome/src/WebRequest.cpp:210: first defined here
c:/users/pinom/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\lib0c8\libESPAsyncWebServer.a(WebRequest.cpp.o): in function `AsyncWebServerRequest::_onTimeout(unsigned int)':
E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer/src/WebRequest.cpp:189: multiple definition of `AsyncWebServerRequest::_onTimeout(unsigned int)'; .pio\build\esp32doit-devkit-v1\lib9c8\libESPAsyncWebServer-esphome.a(WebRequest.cpp.o):E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer-esphome/src/WebRequest.cpp:214: first defined here
c:/users/pinom/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\lib0c8\libESPAsyncWebServer.a(WebRequest.cpp.o): in function `AsyncWebServerRequest::onDisconnect(std::function<void ()>)':
......
......
[ElegantOTA issues.txt](https://github.com/user-attachments/files/17792188/ElegantOTA.issues.txt)

E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer/src/WebAuthentication.cpp:118: multiple definition of `generateDigestHash(char const*, char const*, char const*)'; .pio\build\esp32doit-devkit-v1\lib9c8\libESPAsyncWebServer-esphome.a(WebAuthentication.cpp.o):E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer-esphome/src/WebAuthentication.cpp:112: first defined here
c:/users/pinom/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\lib0c8\libESPAsyncWebServer.a(WebAuthentication.cpp.o): in function `checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*)':
E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer/src/WebAuthentication.cpp:141: multiple definition of `checkDigestAuthentication(char const*, char const*, char const*, char const*, char const*, bool, char const*, char const*, char const*)'; .pio\build\esp32doit-devkit-v1\lib9c8\libESPAsyncWebServer-esphome.a(WebAuthentication.cpp.o):E:\Descargas\Google Drive\DESARROLLO\PRUEBAS\Prueba OTA ESP32/.pio/libdeps/esp32doit-devkit-v1/ESPAsyncWebServer-esphome/src/WebAuthentication.cpp:144: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32doit-devkit-v1\firmware.elf] Error 1
========================================================================================= [FAILED] Took 15.46 seconds =========================================================================================

 *  The terminal process "C:\Users\pinom\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

pinomarcelo, 2024-12-12:
@euphi thanks a lot! It is working now.

BenjaminDerProgrammierer, 2025-01-12:
It isn't a real solution to just downgrade the version, someone should fix it... Although it works...

Just commenting to remove Stale status

closed 2025-01-13 by ayushsharma82