ayushsharma82/ElegantOTA #238: Static IP in ElegantOTA AsyncDemo

renegat56, 2024-12-12:
ElegantOTA works pretty well with DHCP. But I have to set a static IP in my wlan net due to several reasons.
So I inserted :

  // config for static ip
  WiFi.config(myip,gateway,subnet,dns1,dns2);

after :

WiFi.mode(WIFI_STA)

and before :

WiFi.begin(ssid, password)

like usually.

But with ElegantOTA it does not work. I can reach the ESP with http://ip and get the info :

"Hi! This is ElegantOTA AsyncDemo."

But not with http://ip/update. And after trying this, http://ip does not work anymore too.

How can I set static IP? With the deprecated old version I got it working with static IP

Thanks!

mathieucarbou, 2024-12-12:
FYI I am using ElegantOTA in an app and using static IP on ETH and WiFi.
So I know it works ;-)

mathieucarbou, 2024-12-12:
I do exactly like you so this might be something else...
make sure to call begin() on the server AFTER the network stack is ready.

renegat56, 2024-12-12:
So you were right Mathieu.
It was something different. My DHCP server bind9 made the problem. In his files the ESP was registered with the DHCP IP. After I changed in those files the IP to my static IP ElegantOTA worked with the static IP.
Thx for your help!

closed 2024-12-12 by renegat56