ayushsharma82/ElegantOTA #240: MD5 Checksum Validation does not work

rainerfritz, 2024-12-16:
Hi!

I tested the MD5 checksum verification and it has no effect when providing a false checksum digest string.

I could solve it by setting the checksum after Update.begin. 

`#elif defined(ESP32)  
        if (!Update.begin(UPDATE_SIZE_UNKNOWN, mode == OTA_MODE_FILESYSTEM ? U_SPIFFS : U_FLASH)) {
          Serial.print("Failed to start update process\n");
          // Save error to string
          StreamString str;
          Update.printError(str);
          _update_error_str = str.c_str();
          _update_error_str.concat("\n");
          Serial.println(_update_error_str.c_str());
        }     
        // Get file MD5 hash from arg
        if (request->hasParam("hash")) {
          String hash = request->getParam("hash")->value();
          Serial.print(String("MD5 from client: "+hash+"\n").c_str());
          if (!Update.setMD5(hash.c_str())) {
            Serial.print("ERROR: MD5 hash not valid\n");
            return request->send(400, "text/plain", "MD5 parameter invalid");
          }
        }   
      #endif`

ayushsharma82, 2024-12-22:
Interesting... ElegantOTA uses direct API from relevant MCU's Arduino Core so this might be an issue which should be opened on their repository if `Update.setMD5` function is not working properly but please provide complete details for proper diagnosis:

1. Microcontroller:
2. Installed Arduino Core Version:
3. Are you using Arduino IDE or PIO? : 
4. Are you using AsyncWebServer? (ie. Async mode in ElegantOTA) :

rainerfritz, 2024-12-22:
I am using the AsynWebserver on a Heltec Wifi-Lora32 v2 board. 
Arduino Core is 3.0.5 from Tasmota: [Tasmota Espressif](https://github.com/tasmota/platform-espressif32)
using it on Pio. The Update.h is here:
[Update.h](https://github.com/tasmota/arduino-esp32/blob/main/libraries/Update/src/Update.h)

I provided with the call of /ota/start a valid but wrong hash as parameter. When writing finishes, it does not raise the checksum mismatch error. 
When I modify your code that Update.setMD5() is called after Update.begin() is called, the error triggers.

BR
Rainer

Faariq-Siddiqui, 2024-12-29:
@ayushsharma82 I too am facing the same problem.

I am a student using your library for updating my ESP32 Code. 

I am using PlatformIO on VSCode and Arduino Core.

The board is NodeMCU 32-S having a WROOM32 Chip.

I tried to run your code. The one in Demo. The AsyncWebServer Demo one using this falg:

`#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1`

Please help me as well good sir

closed 2025-02-03 by github-actions[bot]