ayushsharma82/ElegantOTA #243: Feedback for “Installation”

StefT09, 2025-01-05:
Hello, I am trying to install ElegantOTA on a NodeMcu V2 but I have a compilation problem:

platformio.ini :

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_compat_mode = strict
lib_deps = ayushsharma82/ElegantOTA@^3.1.6

Build error :

In file included from .pio\libdeps\nodemcuv2\ESPAsyncWebServer\src\AsyncJson.cpp:1:0:
.pio\libdeps\nodemcuv2\ESPAsyncWebServer\src\AsyncJson.h:38:18: error: missing binary operator before token "("
 #if __has_include("ArduinoJson.h")
                  ^>> missing binary operator
In file included from .pio\libdeps\nodemcuv2\ESPAsyncWebServer\src\AsyncMessagePack.cpp:1:0:
.pio\libdeps\nodemcuv2\ESPAsyncWebServer\src\AsyncMessagePack.h:24:18: error: missing binary operator before token "("
 #if __has_include("ArduinoJson.h")
                  ^>> missing binary operator

Thank you for your help

mathieucarbou, 2025-01-05:
I am using:

```
framework = arduino
platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.09.30/platform-espressif32.zip

lib_compat_mode = strict
lib_ldf_mode = chain
lib_deps =
  mathieucarbou/AsyncTCP @ 3.3.1
  mathieucarbou/ESPAsyncWebServer @ 3.4.5
  ayushsharma82/ElegantOTA @ 3.1.6

build_flags =
  ; AsyncTCP
  -D CONFIG_ASYNC_TCP_MAX_ACK_TIME=5000
  -D CONFIG_ASYNC_TCP_PRIORITY=10
  -D CONFIG_ASYNC_TCP_QUEUE_SIZE=64
  -D CONFIG_ASYNC_TCP_RUNNING_CORE=1
  -D CONFIG_ASYNC_TCP_STACK_SIZE=4096
  ; ESPAsyncWebServer
  -D WS_MAX_QUEUED_MESSAGES=64
  ; ElegantOTA
  -D ELEGANTOTA_USE_ASYNC_WEBSERVER=1
  ; C++
  -Wall -Wextra
  -std=c++17
  -std=gnu++17
  -Os
build_unflags =
    -std=gnu++11
```

Example: https://github.com/mathieucarbou/MycilaSafeBoot

mathieucarbou, 2025-01-05:
Sorry! I missed the fact that you are on 8266!

mathieucarbou, 2025-01-05:
```
framework = arduino
platform = espressif8266

lib_compat_mode = strict
lib_ldf_mode = chain
lib_deps =
  mathieucarbou/ESPAsyncWebServer @ 3.4.5
  ayushsharma82/ElegantOTA @ 3.1.6

build_flags =
  ; ElegantOTA
  -D ELEGANTOTA_USE_ASYNC_WEBSERVER=1
  ; C++
  -Wall -Wextra
  -std=c++17
  -std=gnu++17
  -Os
build_unflags =
    -std=gnu++11
```

Example: https://github.com/mathieucarbou/MycilaESPConnect/blob/main/platformio.ini#L11-L46

StefT09, 2025-01-05:
Thank you Mathieu !

(I don't understand much about compilation options, I was getting by on Arduino IDE, I wanted to learn how to use Platformio on a project... it's hard!)

I still have some trouble : 
- with the Example ElegantOTA Example: 
ElegantOTA.begin(&server); // Start ElegantOTA
& is underlined (l'argument de type "ESP8266WebServer *" (aka "esp8266webserver::ESP8266WebServerTemplate<WiFiServer> *") est incompatible avec le paramètre de type "AsyncWebServer *"C/C++(167) )

- on Building
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer @ 3.4.5
|-- ElegantOTA @ 3.1.6
|-- ESP8266WebServer @ 1.0
|-- ESP8266WiFi @ 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Compiling .pio\build\nodemcuv2\lib0f8\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pio\build\nodemcuv2\lib0f8\ESP8266WiFi\CertStoreBearSSL.cpp.o
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
Compiling .pio\build\nodemcuv2\lib0f8\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pio\build\nodemcuv2\lib0f8\ESP8266WiFi\ESP8266WiFiAP.cpp.o
*** [.pio\build\nodemcuv2\src\main.cpp.o] Error 1
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
*** [.pio\build\nodemcuv2\lib0f8\ESP8266WiFi\BearSSLHelpers.cpp.o] Error 1
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
*** [.pio\build\nodemcuv2\lib0f8\ESP8266WiFi\CertStoreBearSSL.cpp.o] Error 1
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
*** [.pio\build\nodemcuv2\lib0f8\ESP8266WiFi\ESP8266WiFi.cpp.o] Error 1
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=c++17'
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
*** [.pio\build\nodemcuv2\lib0f8\ESP8266WiFi\ESP8266WiFiAP.cpp.o] Error 1

closed 2025-02-10 by github-actions[bot]