ayushsharma82/ElegantOTA #249: Authentication Issue

mikecortis, 2025-02-23:
Hello, I am utilising `ElegantOTA.setAuth("username","password");` as given in https://docs.elegantota.pro/security/authentication, and managed to upload it to ESP32. 
Concurrently, I've managed to upload other *.bin without setting the same above username/password.
May be I miss understood this function of setAuth, but I was hoping it would prevent unsolicited uploads on the ESP32 unless the correct username/password is set using ElegantOTA.setAuth.
Can you kindly clarify.

mathieucarbou, 2025-02-23:
> Hello, I am utilising `ElegantOTA.setAuth("username","password");` as given in https://docs.elegantota.pro/security/authentication, and managed to upload it to ESP32. Concurrently, I've managed to [...]

Maybe try with curl or add the logging middleware to see the incoming request headers. 
My assumption is that the browser (or client you use) is caching the username / password and reuses it.

mikecortis, 2025-02-23:
can you pls share an example how to upload using curl? thanks

mikecortis, 2025-02-23:
Could be that EEPROM is conflicting with Auth. in my script, I am saving data on the ESP32 flash utilizing the header EEPROM.h?

mikecortis, 2025-02-23:
i found the issue. i was declaring `ElegantOTA.setAuth` before `ElegantOTA.begin(&server)` hence why.

closed 2025-02-23 by mikecortis