ayushsharma82/ElegantOTA #250: ElegantOTA crash on M5StickCplus 1.1

JK40105, 2025-02-25:
I am using ElegantOTA successfully on several Firebeetle ESP32 and ESP32-E devices.
I use just the few ElegantOTA lines of code (Async) as given in the example.
However when using it on M5StickCPlus 1.1 I get the following error:
12:23:38.023 -> OTA update started!
12:23:38.058 -> OTA Progress Current: 795 bytes, Final: 1150644 bytes
12:23:38.058 -> 
12:23:38.058 -> abort() was called at PC 0x40081a23 on core 1
12:23:38.058 -> 
12:23:38.058 -> 
12:23:38.058 -> Backtrace: 0x400825d1:0x3ffd6af0 0x4008e355:0x3ffd6b10 0x400947bd:0x3ffd6b30 0x40081a23:0x3ffd6bb0 0x400ebed0:0x3ffd6c00 0x400e3821:0x3ffd6c20 0x400e3643:0x3ffd6c40 0x400e37bc:0x3ffd6c60 0x400e29be:0x3ffd6c90 0x4017bd5f:0x3ffd6d00 0x4017bf09:0x3ffd6d50 0x400dd33d:0x3ffd6d80 0x400dddeb:0x3ffd6e00 0x400ddf2d:0x3ffd6e20 0x400d7655:0x3ffd6e40 0x400d76c9:0x3ffd6e70 0x400d805a:0x3ffd6e90 0x4008f0d2:0x3ffd6ec0
12:23:38.106 -> 
12:23:38.106 -> 
12:23:38.106 -> 
12:23:38.106 -> 
12:23:38.106 -> ELF file SHA256: e3ee88cec
12:23:38.106 -> 
12:23:38.359 -> Rebooting...
12:23:38.359 -> ets Jun  8 2016 00:22:57
12:23:38.359 -> 
12:23:38.359 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:23:38.359 -> configsip: 188777542, SPIWP:0xee
12:23:38.359 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:23:38.359 -> mode:DIO, clock div:1
12:23:38.359 -> load:0x3fff0030,len:4916
12:23:38.359 -> load:0x40078000,len:16492
12:23:38.359 -> load:0x40080400,len:4
12:23:38.359 -> load:0x40080404,len:3524
12:23:38.404 -> entry 0x400805b8
12:23:38.801 -> M5StickCPlus initializing...OK

Any help to resolve would be gracefully accepted.

mathieucarbou, 2025-02-25:
This is impossible to help if you do not decode your error: https://maximeborges.github.io/esp-stacktrace-decoder/

ayushsharma82, 2025-02-25:
@mathieucarbou It looks like some misconfigured partitions error?

@JK40105  Please make sure you are choosing the correct "board" in Arduino IDE. (on a second note, if the chosen board is correct then make sure to power cycle once after uploading program via USB/Serial to make them ready for OTA).

ayushsharma82, 2025-02-25:
@JK40105  It also might be that M5Stack board doesn't have the "OTA" partition. You'll have to do some digging.

JK40105, 2025-02-26:
I found out that the M5StickCPlus had been formatted without OTA partition. I do not know how or when this happened, I tried several packages and also UIFlow2, so one of those might have changed the partition scheme.
Many thanks for the support!

closed 2025-03-02 by ayushsharma82