ayushsharma82/ElegantOTA #251: Feature Request for ElegantOTA: Pre-Flash Device Compatibility Validation

inF1704, 2025-02-27:
**## Description**
Currently, ElegantOTA Pro lacks a mechanism to validate whether a firmware binary (.bin) file is compatible with the specific device before flashing begins. This creates a risk where users might accidentally upload firmware intended for a different device model, potentially bricking the device or causing unexpected behavior.

**## Proposed Feature**
Add functionality to ElegantOTA that would allow validating if an uploaded firmware is compatible with the target device before the actual flashing process begins.

**## Use Case**
In environments with multiple similar ESP32/ESP8266 devices running different applications, users might accidentally upload the wrong firmware. For example, uploading firmware intended for a temperature sensor to a device that controls lighting, even though both use the same hardware platform.

**## Potential Implementation Considerations**
There are several approaches that could be considered for this feature:

1. **Filename-based validation**: Allow checking the uploaded firmware's filename against device-specific patterns or identifiers.

2. **Enhanced callback system**: Extend the existing callback system to provide information about the uploaded file before flashing begins.

3. **User confirmation dialog**: Implement a confirmation step in the web interface that shows device information and asks users to confirm compatibility.

4. **Device identifier system**: Allow setting device identifiers in code that could be matched against naming conventions or metadata.

**## Benefits**
- Prevents accidental flashing of incompatible firmware
- Reduces the risk of bricking devices during OTA updates
- Improves user experience by providing clear feedback about compatibility issues
- Particularly valuable in environments with multiple similar device types

The specific implementation approach could be determined based on what's most feasible within the ElegantOTA architecture while maintaining compatibility across different build systems like Arduino IDE and PlatformIO.

ayushsharma82, 2025-02-27:
@inF1704 I believe it's already implemented by espressif deep down. They have magic bytes appended to every bin file to know the compatibility.

Not sure if the 'Update' class in Arduino SDK verifies that. Let me know if you find something about it.

@me-no-dev might help us out here as he has better knowledge of esp32-arduino SDK.

mathieucarbou, 2025-02-27:
@inF1704 : I suggest you use ArduinoOTA to validate. If you find the same issues, then maybe redirect your issue / feature request to Arduino Core. ElegantOTA is just a wrapper around the Arduino Update class also used in ArduinoOTA.

If you need custom user validation (i.e. esp8266 vs esp32, etc), then you might need to first think about how it can be implemented outside of ElegantOTA, so with Update class or ArduinoOTA directly, and propose a common API these libs could use.

closed 2025-04-06 by github-actions[bot]