ayushsharma82/ElegantOTA #256: Upload progress wrongly shown on serial monitor

dariods8474, 2025-03-24:
ElegantOTA shows the update progress in the webpage. I wanted it to also show on the device GLCD screen. So I used this callback function to get info
`void onOTAProgress(size_t current, size_t final) {
  // Log every 1 second
  if (millis() - ota_progress_millis > 1000) {
    ota_progress_millis = millis();
    Serial.printf("OTA Progress Current: %u bytes, Final: %u bytes\n", current, final);
  }
}`
When I use ElegantOTA demo.ino, this is what is seen on the serial monitor
`OTA update started!
Update Received: firmware.bin
OTA Progress Current: 1436 bytes, Final: 0 bytes
OTA Progress Current: 172320 bytes, Final: 170884 bytes
OTA Progress Current: 348948 bytes, Final: 347512 bytes
OTA Progress Current: 528448 bytes, Final: 527012 bytes
OTA Progress Current: 709384 bytes, Final: 707948 bytes
OTA Progress Current: 868780 bytes, Final: 867344 bytes
OTA update finished successfully!` 
The value of variable final being zero throws error restarting the device and the current value being greater than final value does not give real time firmware update progress.
If I used the AsyncDemo.ino the serial monitor shows
`OTA update started!
Update Received: firmware.bin
OTA Progress Current: 1436 bytes, Final: 868780 bytes
OTA Progress Current: 172320 bytes, Final: 868780 bytes
OTA Progress Current: 348948 bytes, Final: 868780 bytes
OTA Progress Current: 528448 bytes, Final: 868780 bytes
OTA Progress Current: 709384 bytes, Final: 868780 bytes
OTA Progress Current: 867344 bytes, Final: 868780 bytes
OTA update finished successfully!`
Surprisingly the update progress is shown properly in the webpage for both the example files

NDC-SC, 2025-04-07:
I get exactly the same problem with the Final parameter increasing at each step (0, 103k, 273k...850k), rather than being fixed at the binary file size.

void OTAprogress(size_t current, size_t final) {
  if (millis() - OTAlag > 1000) {
    Serial.print("OTA progress "); Serial.print(current); Serial.print(" "); Serial.println(final);
    OTAlag = millis();
}}

OTA update started.
Update Received: Listing1-1.ino.bin
OTA progress 1436 0
OTA progress 104828 103392
OTA progress 274276 272840
OTA progress 463828 462392
OTA progress 660560 659124
OTA progress 851548 850112
OTA update finished

Can you please provide the instruction to obtain the binary file size ?
Then a work around is (current parameter x 100)/binary file size.

Thanks

dariods8474, 2025-04-07:
Follow the the instructions in AsyncDemo.ino which uses EspAsyncWebServe instead of Demo.ino that uses WebServer library. I use that and I have no problems. I am lucky that this problem took place. Now I am able to/using ElegantOTA, WebSerial and GPIO viewer which all need EspAsyncWebServer. Do Check out EspAsyncWiFiMansger https://github.com/alanswx/ESPAsyncWiFiManager.git

NDC-SC, 2025-04-07:
Rather than use library B to resolve an issue with library A, it would be helpful to understand the cause of the issue with the latter.
Hopefully the ElegantOTA author can help.

dariods8474, 2025-04-09:
The author is more interested in selling pro version than responding here to queries

SuperNinja73, 2025-04-26:
> The author is more interested in selling pro version than responding here to queries

yes but if the bug is also present in the pro version, he might be interested in it

ayushsharma82, 2025-04-26:
Hello guys,

Please don't take my unresponsiveness the wrong way. In last few months my focus has been on some other important things in my life (before you say it - no, not the pro libraries 😄); hence the reason for not responding to every issue on Github.

The reason is because ElegantOTA uses `HTTPUpload` class of Arduino ESP32 core in sync mode to handle uploads and `HTTPUpload` is not properly parsing the `totalSize` data. Reference line:  https://github.com/ayushsharma82/ElegantOTA/blob/master/src/ElegantOTA.cpp#L312

There is very little which I can do in ElegantOTA to mitigate this as it is an issue in external dependency and I doubt it will be debugged quickly as this bug is very dependent on user's setup (OS, browser, firewall etc.).

Feel free to open an issue in ESP32 Arduino Core (https://github.com/espressif/arduino-esp32) about this or debug it if possible.

dariods8474, 2025-04-27:
If this bug is dependent on the user setup(OS, browser, firewall, etc) how does the update progress show properly in the browser and when using Async WebServer the browser and serial monitor shows the right file upload progress. This bug is related to ElegantOTA source code not Arduino ide

ayushsharma82, 2025-04-27:
The progress you see on the webpage is a "client-side" implementation, that's not being sent back by the ESP32 to your browser; instead it's calculated on-the-fly in the browser itself because 1) it knows how large the file is to begin with and 2) how many bytes the XHR request has already sent.

The progress you get inside your firmware is based on HTTP server implementation (`HTTPUpload`) functioning correctly.

closed 2026-09-05 by ayushsharma82