ayushsharma82/ElegantOTA #257: Support RP2040 with W5100

ZinnerC, 2025-03-24:
Hi would really like to use ElegantOTA in one of my projects where I use a W5100 chip to have network connection.
It was possible to just skip the "__isPicoW" part and have it compiled for RP2040+W5100.

Would it be possible to have an other check instead of the "__isPico" in 

```
ElegantOTAClass::begin(...)
...
#if defined(TARGET_RP2040) || defined(TARGET_RP2350) || defined(PICO_RP2040) || defined(PICO_RP2350)
    // DISABLED by CAZ
    // if (__isPicoW) {
    //   ELEGANTOTA_DEBUG_MSG("RP2040: Not a Pico W, skipping OTA setup\n");
    //   return;
    // }
  #endif
...
```

ayushsharma82, 2025-03-27:
Hi @ZinnerC ,

Sure. Which library do you use for W5100? I'll see if it has any internal directive which we can detect in ElegantOTA.

ZinnerC, 2025-04-01:
I use "W5100lwIP.h" which is included in https://github.com/earlephilhower/arduino-pico/tree/master/libraries/lwIP_w5100