ayushsharma82/ElegantOTA #260: SPFIFFS not replaced correctly

Owlbertz, 2025-04-21:
Hi,

I am using this library to update the actual firmware of my ESP32, which always worked without problems. Now I needed do modify the SPIFFS filesystem as well, but uploading a new build resulted in the filesystem being somehow broken.

`SPIFFS.usedBytes(), SPIFFS.totalBytes()` shows `"0.000000/1.832760"` with the filesystem binary being build having 1.4 MB.

Is there any way to fix this OTA or do I need to break apart my installation to attach the ESP to my PC for a reset?

ayushsharma82, 2025-04-26:
Hi @Owlbertz , Did you use a different flash partition layout this time? This happens only when the partition size is different than what was initially used or when the network causes data corruption.

Unlike firmware uploads which keep a separate empty "ota" partition for download, SPIFFS updates overwrite the content as the ESP receives it, so doing everything systematically is important.

You can try uploading the SPIFFS image with and without different partition layout to see if it's recoverable.

Owlbertz, 2025-04-29:
Hi @ayushsharma82, thanks for the response. I am not sure what I did wrong, but I went ahead and just retrieved the device and deploy the filesystem via USB cable. This resolved the issue and all additional filesystem updates using the OTA libraray worked fine. Maybe I tried to push a corrupt binary one time causing the partition to become corrupt.

closed 2025-04-29 by Owlbertz