ayushsharma82/ElegantOTA #261: Following Random Nerd Tutorial creates fragile code usage....

RobertLRead, 2025-04-23:
Dear ElegantOTA developers (Ayush Sharma),

Thank you for your nice software. We have followed a tutorial from "Random Nerd Tutorials" [https://randomnerdtutorials.com/esp32-ota-elegantota-arduino/](https://randomnerdtutorials.com/esp32-ota-elegantota-arduino/), which has allowed us to use this library on an ESP32.

However, that tutorial requires us to make a change directly in your header file.  That tutorial requires:

[image]

Is there a way that we can accomplish this change without having to modify your code, which of course is hard to repeat for our developers?

Is there a simpler way for us to solve this problem?  We are using version 3.1.6 for an ESP32 Dev Module board compile in the Arduino IDE.

RobertLRead, 2025-04-23:
We have seen that this issue has come up in the "Closed issues", but still this makes it hard to use this fine software as an independent hassle-free library.  Novice programmers will not know how to do this, and without the example of Random Nerd Tutorials, we would not have succeeded using your library.

I suggest that this could be made a switch passed into the "begin" method of the class. This could be done in a back-compatible way by putting a new argument at the end of the list of the parameters to "begin" and providing it a default.

mathieucarbou, 2025-04-23:
Setting compilation flags is an IDE issue, not a library issue. This lib supports that.

I would encourage you to look at Platformio tooling and pioarduino framework instead of using Arduino IDE. 

At least, you will get a real dev environment with correct dependency management also.

Arduino IDE is a toy, good for PoC only.

RobertLRead, 2025-04-23:
Thank you. We use PlatformIO for larger projects. Nonetheless, there is tremendous value in being able to make such useful code available to people using the Arduino IDE---professional programmers are made from novice programmers who are exposed to the power of good software. Being able to produce an Over-the-Air Proof of Concept seems somewhat valuable. Nonetheless, I understand you may wish to close this issue without taking further action.

mathieucarbou, 2025-04-23:
> Thank you. We use PlatformIO for larger projects. Nonetheless, there is tremendous value in being able to make such useful code available to people using the Arduino IDE---professional programmers [...]

the library works without this flag. (with the default Arduino WebServer).

this flag is just for people who wants to use ESPAsyncWebServer and need an alternative implementation to be activated.

You might want to read the doc online for more details.

closed 2025-05-31 by github-actions[bot]