ayushsharma82/ElegantOTA #268: Captive Portal and ElegantOTA

bobskigit, 2025-06-17:
The combination of a Captive Portal and ElegantOTA doesn't seem to work.

There are no errors in the serial monitor, and I can see the /update request, but then nothing happens.

All of the captive HTML pages are served from SPIFFS.

Any suggestions?

StefanoTesla, 2025-06-26:
Hi Bob, I was opening a similar question not related to OTA, but the implementation of the async web server.

In the library the async web server is a dependency:

  "dependencies": [
    {
      "owner": "esp32async",
      "name": "ESPAsyncWebServer",
      "version": "^3.7.3",
      "platforms": ["espressif8266", "espressif32", "raspberrypi"]
    }
  ],

the problem is the ^ on the version, try to modify the library.json of elegant ota to force the usage of version 3.7.7 and I hope you will solve the problem.

I have the same problem with the WiFiManager AP web pages..

bobskigit, 2025-06-26:
If you exclude certain URL's from the captive portal handler it works.

 bool canHandle(__unused AsyncWebServerRequest *request) const override
  {
    // Only handle requests that are not for the update or OTA endpoints
    // This is to avoid conflicts with OTA updates and firmware updates
    return request->url() != "/update" && request->url() != "/ota/start" && request->url() != "/ota/upload";
  }